Lora-net/SWL2001 #5: Makefile ignores MCU_FLAGS in v3.2.4

adriansmares, 2022-10-22:
I am used to providing build parameters via [direnv](https://direnv.net/). It took me a while to understand why `MCU_FLAGS` does not work unless I add it after the build targets:
https://github.com/Lora-net/SWL2001/blob/8d052b21f39344a543c15e29d76dfab2cc9eb709/Makefile#L36

This should've been `?=`, not `=?`.

matthieuantoine, 2022-10-23:
Thanks @adriansmares - good catch. I'll report that internally to fix it for the next release.

lbm-team, 2023-03-06:
Hi @adriansmares !
Thanks for your report.
We confirm this will be corrected in next release

lbm-team, 2023-06-29:
Hi @adriansmares !

This issue is now fixed in new v3.3.0 release and mentioned in [CHANGELOG](https://github.com/Lora-net/SWL2001/blob/v3.3.0/CHANGELOG.md).   

Line updated in `Makefile` can be found [here](https://github.com/Lora-net/SWL2001/blob/3258275fbceb02090300cac8cf651d00fb418e0b/Makefile#L36)

Once more thanks for your report

closed 2023-06-29 by lbm-team