Lora-net/SWL2001 #6: Example application overrides EUIs while using LR11xx crypto

adriansmares, 2022-10-22:
The example application overrides the DevEUI/JoinEUI and AppKey always on reset:
https://github.com/Lora-net/SWL2001/blob/0979db31e5daedbcd7bef20d699dfda1efa6b307/utilities/user_app/main_examples/main_exti.c#L200-L211

This should be a bit more nuanced, since if the example is compiled with `CRYPTO=LR11XX_WITH_CREDENTIALS`, one may expect that the EUIs are really the ones 'on the board'.

matthieuantoine, 2022-10-23:
Thanks @adriansmares - I'll report that internally to update it for the next release.

FYI - examples in the [LoRa Basics Modem SDK](https://github.com/Lora-net/SWSD001) already have it implemented this way.

lbm-team, 2023-03-06:
Hi @adriansmares !
Thanks for your report.
We will update the basic example to take in count internal credentials in case `CRYPTO=LR11XX_WITH_CREDENTIALS` is used at makefile level.  
It can also make sense to print current CHIP EUI and CHIP PIN to ease claiming process.  
It will be updated as well for the next release

lbm-team, 2023-06-29:
Hi @adriansmares !

We add this feature in the new v3.3.0 release.   
You can find the updates line in main example [here](https://github.com/Lora-net/SWL2001/blob/3258275fbceb02090300cac8cf651d00fb418e0b/utilities/user_app/main_examples/main_exti.c#L290C8-L290C8)

closed 2023-06-29 by lbm-team