Lora-net/SWL2001 #9: Size Error in smtc_secure_element_get_pin()

rdaubencd, 2022-12-13:
There is a possible size error in smtc_modem_core\smtc_modem_crypto\soft_secure_element\soft_se.c:smtc_secure_element_get_pin().  The memcpy uses a size of SMTC_SE_EUI_SIZE, but probably should use SMTC_SE_PIN_SIZE.

lbm-team, 2023-03-06:
Hi @rdaubencd !
Thanks for the catch!
It will be corrected for the next release

lbm-team, 2023-06-29:
Hi @rdaubencd !  
This issue is now solved in the new v3.3.0 release of the project.  
The fix is mentioned in [CHANGELOG](https://github.com/Lora-net/SWL2001/blob/v3.3.0/CHANGELOG.md).  
You can also see the fix in code [here](https://github.com/Lora-net/SWL2001/blob/3258275fbceb02090300cac8cf651d00fb418e0b/smtc_modem_core/smtc_modem_crypto/soft_secure_element/soft_se.c#L655)

closed 2023-06-29 by lbm-team