Lora-net/SWL2001 #10: No events from the time synchronization using MAC sync

RobTWTG, 2023-06-16:
LoRa Basics Modem version: 3.1.7

In my project, we have integrated this stack and are currently producing products. While updating a new feature that involves time synchronization, I encountered the following issue:

When I call the function smtc_modem_time_start_sync_service(STACK_ID, SMTC_MODEM_TIME_MAC_SYNC);, there is no callback triggered with current_event.event_type == SMTC_MODEM_EVENT_TIME. Additionally, setting a small interval using the smtc_modem_time_set_sync_interval_s function does not seem to have any effect.

Although I can receive the correct time when polling with the smtc_modem_get_time function, no callbacks are triggered when a synchronization exchange has occurred.
I was wondering if you can help us fix this issue.

martinichka, 2023-06-19:
@matthieuantoine this is the same as emailed you before. Can you reproduce the problem?

lbm-team, 2023-06-21:
Hi @RobTWTG @martinichka ,

We have not seen this issue at our side during our v3.1.7 validation. 
Do you have any modem traces that can show what happened after the start of the time sync service?

Regarding the change of the interval between sync, could you please call the function just before starting the time sync service so that it can be taken in count directly and not after next synchronization event (36h).

If possible in your development environment/project could you please try the v3.2.4 version of the library to see if it can solve your issue (we have modify a bit this service between this 2 releases)?

Let me know if it helps
Audrey from lbm-team

RobTWTG, 2023-07-04:
While setting up the firmware to get modem traces, I found out this was an issue on our side by calling the lr1mac_core_init function an extra time. This function resets the device_time_callback and context back to NULL pointers.

Thank you for the reply, I will close the issue.