Lora-net/SWL2001 #11: Invalid MIC on Relay Feature

EduardoPfeifer, 2023-07-31:
Hey!
I am testing the new branch (feature/relay) with the ChirpStack network server and I am getting some issues after the device is successfully connected to the NS through the Relay RX device.

After a `TX DONE on RXR` (Join Accept sent to the device), the Relay RX starts a CAD 1s.

The device, after a Join accept, starts an Alarm to send uplink in 60 s. When the device sends the WOR, the Relay tells me a
`Invalid MIC` and `Unknown ED (0x017501A3)`.

What am I doing wrong? Is this behavior expected or could it be a possible bug?

Thanks!

EduardoPfeifer, 2023-08-01:
Hi!

I found the problem, it was my setting on Chirpstack about `ForwardListReq`.

closed 2023-08-01 by EduardoPfeifer

Mathannbala, 2024-03-15:
Hi @EduardoPfeifer, could you be kind enough to clarify more on the settings of Chirpstack with regards to the ForwardListReq? Im encountering the same issue of Invalid MIC and Unknown ED. As i do not own the Chirpstack server, i would need to raise it to the owners for the appropriate changes. Hope you can assist me on this. Thanks
[image]
[image]

lbm-team, 2024-03-15:
Hi @Mathannbala , 

In Chirpstack, you need to link your end-device and your relay together

To do that,select you ED  in your application , click on "selected device" then "Add to relay"
[image]

I hope it help

Mathannbala, 2024-03-15:
@lbm-team Thank you so much for the prompt response, I've done so and my end device is able to connect via the relay. However, I noticed that after a certain amount of transmissions, I'm seeing " missed data" on the relay, not sure if this is expected behaviour