Lora-net/SWL2001 #12: Driver function lr_fhss_get_hop_sequence_count() returns a status instead of a count.

rdaubencd, 2023-08-11:
This driver function is called with the expectation that it will return an unsigned integer for the count.  But its interface is defined with a return of a status.

From ral_sx126x.c this is the implementation:
`ral_status_t ral_sx126x_lr_fhss_get_hop_sequence_count( const void*                 context,
                                                        const ral_lr_fhss_params_t* lr_fhss_params )
{
    sx126x_lr_fhss_params_t sx126x_params;
    ral_sx126x_convert_lr_fhss_params_from_ral( lr_fhss_params, &sx126x_params );

    return ( ral_status_t ) sx126x_lr_fhss_get_hop_sequence_count( &sx126x_params );
}`

This should be changed to return the actual count:
`unsigned int ral_sx126x_lr_fhss_get_hop_sequence_count( const void*                 context,
                                                        const ral_lr_fhss_params_t* lr_fhss_params )
{
    sx126x_lr_fhss_params_t sx126x_params;
    ral_sx126x_convert_lr_fhss_params_from_ral( lr_fhss_params, &sx126x_params );

    return sx126x_lr_fhss_get_hop_sequence_count( &sx126x_params );
}`

The other radio drivers probably have the same issue.

lbm-team, 2023-11-14:
Hi @rdaubencd ,  
Thanks for your message.  
Good catch!
This issue will be solved in next LBM release.

lbm-team, 2023-12-18:
Hi @rdaubencd 

This issue is now fixed in new v4.3.0 release and mentioned in [CHANGELOG](https://github.com/Lora-net/SWL2001/blob/v4.3.0/CHANGELOG.md).

Line updated in [ral_sx126x.c](https://github.com/Lora-net/SWL2001/blob/v4.3.0/lbm_lib/smtc_modem_core/smtc_ral/src/ral_sx126x.c) can be found [here](https://github.com/Lora-net/SWL2001/blob/ffca22633d111ef2874fa7c2f35498a57d87f3f0/lbm_lib/smtc_modem_core/smtc_ral/src/ral_sx126x.c#L1045)

Line updated in [ral_lr11xx.c](https://github.com/Lora-net/SWL2001/blame/v4.3.0/lbm_lib/smtc_modem_core/smtc_ral/src/ral_lr11xx.c) can be found [here](https://github.com/Lora-net/SWL2001/blob/ffca22633d111ef2874fa7c2f35498a57d87f3f0/lbm_lib/smtc_modem_core/smtc_ral/src/ral_lr11xx.c#L1055)

Once more thanks for your report

closed 2023-12-18 by lbm-team