Lora-net/SWL2001 #15: Relay MAC commands sent as a response to a relay forward request are ignored

adriansmares, 2023-09-14:
When LBM is compiled with relay RX support, LBM will ignore relay MAC commands (FPort = 0) which are sent as a response to a uplink forwarding request (FPort 226). This happens because the following part of the MAC command parser checks the _TX_ port instead of the _RX_ port:

https://github.com/Lora-net/SWL2001/blob/c4f386ac54b3e35f45b8ba7502207faa6f634921/smtc_modem_core/lr1mac/src/lr1_stack_mac_layer.c#L1424-L1440

I think this is a mistake, as non-relay MAC commands are processed, but relay MAC commands themselves cannot be sent as a response to a relay forwarding request. A possible fixup may be to check the RX metadata instead:

```c
if( ( lr1_mac->rx_down_data.rx_metadata.rx_fport_present == true ) && ( lr1_mac->rx_down_data.rx_metadata.rx_fport == FPORT_RELAY ) )
```

lbm-team, 2023-09-15:
Hello @adriansmares,

Thanks for the catch!
 It will be fixed in the next release. 
In the meantime, your proposal is valid.

lbm-team, 2023-12-22:
Hello,

This issue was solved in the release v4.4.0-feature-relay.
I will let you close the issue if the fix is ok for you.

adriansmares, 2024-01-12:
The fix looks good to me and I cannot reproduce this anymore in `v4.4.0-feature-relay` 👍 .

closed 2024-01-12 by adriansmares