Lora-net/SWL2001 #16: Relay forwarding task is executed 2 seconds after the transmission, instead of 50 milliseconds

adriansmares, 2023-09-15:
When LBM is compiled with relay RX support, the 'relaying' task (which re-transmits the uplink message from the end device) is executed 2 seconds after the transmission end, instead of 50 milliseconds after the transmission end:

https://github.com/Lora-net/SWL2001/blob/c4f386ac54b3e35f45b8ba7502207faa6f634921/smtc_modem_core/modem_services/relay_service/lorawan_relay_rx_service.c#L187-L208

The enqueuing itself is done correctly - it uses the receive timestamp + 50 milliseconds:

https://github.com/Lora-net/SWL2001/blob/c4f386ac54b3e35f45b8ba7502207faa6f634921/smtc_modem_core/lr1mac/src/relay/relay_rx/relay_rx.c#L1299-L1301

I guess this change from the specification is done in order to avoid the relay interfering with the RX1 and RX2 of the original transmission, and to be honest it sort of makes sense (although it assumes RX1=1), but I think that either the delay in the regional parameters should be updated or this delay should be somehow controllable - it makes deduplication on the Network Server significantly harder if we want to keep all downlink paths. (we need a very large window now)

lbm-team, 2023-09-18:
Hello @adriansmares ,

This 2s delay was introduce for some tests and should have not made it to the release. 
This would be fixed in the next release. 

In the meantime, you could remove the "+2" in the `task_relay` variable.

lbm-team, 2023-12-22:
Hello,

This issue was solved in the release v4.4.0-feature-relay.

closed 2023-12-22 by lbm-team