Lora-net/SWL2001 #19: fuota_packges buffers can overflow

MpmTwtg, 2023-09-25:
The ans_index is able to overflow fmp_tx_payload_ans.

https://github.com/Lora-net/SWL2001/blob/45e19e5bf783bf874b8c9825b76c4c5befa8bcba/smtc_modem_core/lorawan_packages/fuota_packages/lorawan_fmp_package.c#L498

The reason behind this is that it loops and can fill the buffer as long ass there are msg's, but the buffer is as big as the biggest package.
https://github.com/Lora-net/SWL2001/blob/45e19e5bf783bf874b8c9825b76c4c5befa8bcba/smtc_modem_core/lorawan_packages/fuota_packages/lorawan_fmp_package.c#L490

https://github.com/Lora-net/SWL2001/blob/45e19e5bf783bf874b8c9825b76c4c5befa8bcba/smtc_modem_core/lorawan_packages/fuota_packages/lorawan_fmp_package.c#L188

This can also happen in
lorawan_fragmentation_package.c
and possible in remote_multicast_tx_payload_ans
The same problems occurs in fuota_packages_v2

lbm-team, 2023-10-04:
Thank you for your feedback, this issue will be fixed in the next release

lbm-team, 2023-12-18:
Hi @MpmTwtg

This issue is now fixed in new v4.3.0 release and mentioned in [CHANGELOG](https://github.com/Lora-net/SWL2001/blob/v4.3.0/CHANGELOG.md).

Once more thanks for your report

closed 2023-12-18 by lbm-team