Lora-net/SWL2001 #22: Remote Multicast Setup command fails

HernandoSergio, 2023-11-02:
https://github.com/Lora-net/SWL2001/blob/45e19e5bf783bf874b8c9825b76c4c5befa8bcba/smtc_modem_core/lorawan_packages/fuota_packages/lorawan_remote_multicast_setup_package.c#L890C1-L897C25

The validation of the frequency and the DR is done wrong, it shoul be checked with "OKLORAWAN" as the api validation functions return "OKLORAWAN" (0) or  "ERRORLORAWAN" (-1) instead of true or false.

lbm-team, 2023-11-14:
Hi @HernandoSergio 
Thanks for your message.
Good catch!
This issue will be solved in next LBM release.

lbm-team, 2023-12-18:
Hi @HernandoSergio

This issue is now fixed in new v4.3.0 release and mentioned in [CHANGELOG](https://github.com/Lora-net/SWL2001/blob/v4.3.0/CHANGELOG.md).

Once more thanks for your report

closed 2023-12-18 by lbm-team