Lora-net/SWL2001 #25: Class C reception active after session end

HernandoSergio, 2023-11-13:
After the end of a Class C multicast session, the device seems to keep the reception active, as the log shows receptions

[image]

lbm-team, 2023-11-14:
Hi @HernandoSergio,
This is an expected behavior: An internal package cannot take the decision to stop the unicast session of class C or class B at the end of a multicast session as the user may has started it at any time.  
To inform the upper layer, the modem will generate 2 dedicated events `SMTC_MODEM_EVENT_NO_MORE_MULTICAST_SESSION_CLASS_C` and `SMTC_MODEM_EVENT_NO_MORE_MULTICAST_SESSION_CLASS_B` so that the modem user can decide whether or not return to class A by calling `smtc_modem_set_class` with `SMTC_MODEM_CLASS_A`

HernandoSergio, 2023-11-14:
Yes, I understand what you say, but the Class C unicast was not active before the multicast session, the device was in Class A.

closed 2024-04-08 by lbm-team