Lora-net/SWL2001 #29: FCntUp does not reset to 0 after join accept is processed by LoRa Basics Modem

atektelic, 2023-12-20:
According to LoRaWAN v1.0.4 specification, section 4.3.1.5 (line 685-686):

_Whenever an OTAA end-device successfully processes a Join-Accept frame, the frame counters on the end-device (FCntUp) and the Network side (FCntDown) are reset to 0_. 

On LoRa Basics Modem, **FCntUp** continuously increments on every uplink (starting from the first join request) and does not reset to zero after join -- which does not align with the specification.

Is this a deliberate design choice? Or perhaps a detail that was missed?

mluis1, 2023-12-21:
A similar question was asked under LoRaMac-node project. Please refer to the following link for the provided answer.
https://github.com/Lora-net/LoRaMac-node/discussions/1571

lbm-team, 2023-12-21:
LBM resets to 0 the FCntUp after each join accept . do you perform a leave before rejoining?

atektelic, 2023-12-21:
Thanks @mluis1! I will close this ticket since your reply addresses our issue.

closed 2023-12-21 by atektelic