Lora-net/SWL2001 #32: LCTT test case 12B fails due to system reset

atektelic, 2024-01-15:
Hello,

During LCTT test case 12B,  a _LinkADRReq_ is sent (to the end device) to set data rate to DR5 (which is used for LR-FHSS in US915). We expect the end device to NACK the _LinkADRReq_ on the _LinkADRAns_, however, the end device resets and eventually fails LCTT test case 12B.

From [this](https://github.com/Lora-net/SWL2001/files/13943517/lctt-12b-crash.log) log file, the reset is caused from calling `region_us_915_lora_dr_to_sf_bw` function (see below), wherein DR5 falls under the `else` clause which runs `smtc_modem_hal_lr1mac_panic` causing the system to reset:
```
void region_us_915_lora_dr_to_sf_bw( uint8_t in_dr, uint8_t* out_sf, lr1mac_bandwidth_t* out_bw )
{
    if( ( in_dr <= DR4 ) || ( ( in_dr >= DR8 ) && ( in_dr <= DR13 ) ) )
    {
        *out_sf = datarates_to_sf_us_915[in_dr];
        *out_bw = datarates_to_bandwidths_us_915[in_dr];
    }
    else
    {
        smtc_modem_hal_lr1mac_panic( );
    }
}
```
It seems, by default, LoRa Basics Modem compiles with `RP2_103` –  this sets DR5 (along with other DRs) as supported DR for US915 during compile time (i.e. enables bitfield for DR5 as supported DR). With DR5 enabled, _LinkADRReq_ command requesting DR=5 passes the valid DR check. But once the sensor tries to transmit, `region_us_915_lora_dr_to_sf_bw` function is called and causes a reset (as explained above).

Our current work around to pass LCTT test case 12B is to compile LoRa Basics Modem with `RP2_101` flag enabled  to use RP002-v1.0.1 specs.

Have we missed anything to prevent this behaviour? Or maybe this is expected? Any input is much appreciated.

Thanks!

lbm-team, 2024-01-16:
Hello atektelic,

In your trace log it appears that traces have been added:  
"UPLINK #401: Port: 2, DR3 (903.10 MHz, SF7, BW125), Payload[1]: 02"

Before this trace, do you check the modulation type with `smtc_real_get_modulation_type_from_datarate()` to use the right conversion of datarate ? `smtc_real_lora_dr_to_sf_bw()` for LoRa or `smtc_real_lr_fhss_dr_to_cr_bw()` for LRFHSS

There is an example of use here: https://github.com/Lora-net/SWL2001/blob/ffca22633d111ef2874fa7c2f35498a57d87f3f0/lbm_lib/smtc_modem_core/lr1mac/src/lr1mac_core.c#L234C10-L234C10

Best

atektelic, 2024-01-16:
Thanks for your input, closing this ticket now as it is a problem on our end.

closed 2024-01-16 by atektelic