Lora-net/SWL2001 #44: Flash porting test always failing due to array out of bounds access

non-det-alle, 2024-03-02:
[This line](https://github.com/Lora-net/SWL2001/blob/ffca22633d111ef2874fa7c2f35498a57d87f3f0/lbm_examples/main_examples/main_porting_tests.c#L1463) in `lbm_examples/main_examples
/main_porting_tests.c` will always cause a SIGSEGV because the `modem_context_type_t` enum element `CONTEXT_SECURE_ELEMENT=4` is used as index of the `name_context_type` array which has max index 3.

lbm-team, 2024-03-04:
Hi non-det-alle,

can you try with this modification:

static const char* name_context_type[] = { "MODEM", "KEY_MODEM", "LORAWAN_STACK",  "FUOTA", "SECURE_ELEMENT", "STORE_AND_FORWARD" };

non-det-alle, 2024-03-04:
Hello. Yes I know, that's what I already proposed in pull request #45 related to this issue. I'll change the names as per your suggestion.