Lora-net/SWL2001 #46: Is there a file named ralf_llcc68.c?

ltmzk, 2024-03-05:
Hello,

I am trying to apply this project to LLCC68 radio chip. However, when I read through `lbm_lib\smtc_modem_core\smtc_ralf\src`, there is no files like ralf_llcc68.c, while in smtc_ral\src, I could see both ral_sx126x.c and ral_llcc68.c. I'm wondering if I need to create ralf_llcc68.c by myself, or just there is no need for ralf_llcc68.c. 

If anybody has any insight, help is greatly appreciated.

Thanks.

lbm-team, 2024-04-08:
Hello,

There is no plan to officially support the llcc68 with LBM.
Best

closed 2024-04-08 by lbm-team

djObsidian, 2025-02-27:
LLCC68 works fine with sx1262 target, I tested it. Just keep in mind supported data rates and SFs are different, so select them appropriately and disable ADR. 
Please note that LLCC68 doesn't support all required LoRaWAN specification data rates and device with it cannot pass LoRa-Alliance certification, but it might be fine for private networks.