Lora-net/SWL2001 #47: cannot get geolocation example to join

rlimberger, 2024-03-05:
Im using the LR1120 evaluation board, updated the tranceiver FW to version 0201 and am trying to run the geolocation example on it. However, i get errors stating `the device is already join`

```
INFO: Modem Initialization                                                      
INFO: Use soft secure element for cryptographic functionalities                 
stack_id 0                                                                      
 DevNonce = 0                                                                   
 JoinNonce = 0xff ff ff, NetID = 0xff ff ff                                     
 Region = US915                                                                 
INFO: LoRaWAN Certification is disabled on stack 0                              
almanac_services_init                                                           
mw_gnss_scan_services_init task_id 8, service_id 0, CURRENT_STACK:0             
mw_gnss_send_services_init task_id 9, service_id 0, CURRENT_STACK:0             
mw_gnss_almanac_services_init task_id 10, service_id 0, CURRENT_STACK:0         
mw_wifi_scan_services_init task_id 11, service_id 0, CURRENT_STACK:0            
mw_wifi_send_services_init task_id 12, service_id 0, CURRENT_STACK:0            
Store and fwd # scanning for filesystem...                                      
Store and fwd # found existing filesystem, usage: 0/225                         
CIRCULARFS read: {0,0} cursor: {0,0} write: {0,0}                               
[0000] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0001] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0002] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0003] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0004] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0005] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0006] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0007] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0008] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
[0009] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEE                    
get_event () callback                                                           
INFO: Event received: RESET                                                     
ERROR: smtc_modem_set_deveui call but the device is already join                
ERROR: smtc_modem_set_joineui call but the device is already join               
ERROR: smtc_modem_set_nwkkey call but the device is already join                
ERROR: smtc_modem_set_region call but the device is already join                
INFO: smtc_modem_join_network                                                   
WARN: smtc_modem_join_network call but the device is already join               
Added task in supervisor for next GNSS almanac status check at 0 + 0 s          
Store and fwd # format filesystem...                                            
mw_wifi_scan_add_task: add task in supervisor at 0 + 0 s                        
mw_gnss_scan_add_task: add task in supervisor at 0 + 0 s                        
INFO: GEOLOCATION example is starting                                           
INFO: LR11XX FW: 0x0201, type: 0x02                                             
Enqueued RP task for Wi-Fi scan (hook_id #3)                                    
INFO: gnss almanac task launched                                                
WARN: No time available: 32K STOPPED                                            
Check almanac status again in 300s                                              
Enqueued RP task (ASAP) for GNSS almanac demodulation (hook_id #1)              
INFO: gnss_almanac_rp_task_done                                                 
ALMANAC task self aborted                                                       
get_event () callback                                                           
INFO: Event received: GNSS_ALMANAC_DEMOD_UPDATE                                 
ALMANAC_DEMOD_UPDATE info:                                                      
-- status GPS: UNKNOWN                                                          
-- status BEIDOU: UNKNOWN                                                       
-- statistics:                                                                  
   nb update from sat done: 0                                                   
   nb update from sat success: 0                                                
   nb aborted by RP: 0                                                          
   cumulative timing: 0 s                                                       
GPS progress: 0%                                                                
BDS progress: 0%                                                                
aborted: 0                                                                      
Added task in supervisor for next GNSS almanac status check at 0 + 300 s        
Enqueued RP task for GNSS scan at 1105 (duration 80 s)                          
INFO: gnss task launch at 1104 for navgroup.scan[0]                             
radio configure for scan done 
```

rlimberger, 2024-03-05:
Here is the error output when running the `periodic_uplink `example:

```
INFO: Modem Initialization                                                      
INFO: Use soft secure element for cryptographic functionalities                 
stack_id 0                                                                      
 DevNonce = 0                                                                   
 JoinNonce = 0xff ff ff, NetID = 0xff ff ff                                     
 Region = US915                                                                 
INFO: LoRaWAN Certification is disabled on stack 0                              
Modem event callback                                                            
INFO: Event received: RESET                                                     
ERROR: smtc_modem_set_deveui call but the device is already join                
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 30
4): SMTC_MODEM_RC_BUSY                                                          
ERROR: smtc_modem_set_joineui call but the device is already join               
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 30
5): SMTC_MODEM_RC_BUSY                                                          
ERROR: smtc_modem_set_appkey call but the device is already join                
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 30
6): SMTC_MODEM_RC_BUSY                                                          
ERROR: smtc_modem_set_nwkkey call but the device is already join                
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 30
7): SMTC_MODEM_RC_BUSY                                                          
ERROR: smtc_modem_set_region call but the device is already join                
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 31
6): SMTC_MODEM_RC_BUSY                                                          
INFO: smtc_modem_join_network                                                   
WARN: smtc_modem_join_network call but the device is already join               
ERROR: In main_examples/main_periodical_uplink.c - modem_event_callback (line 31
8): SMTC_MODEM_RC_BUSY                                                          
INFO: Periodical uplink example is starting 
```

opeyrard, 2024-05-27:
Please use GNU Arm Embedded Toolchain 10-2020-q4-major 10.2.1 20201103 (release) or lower version.
Could you please try and let us know if issue remains ?
Many thanks.

opeyrard, 2024-09-02:
No update, so issue is closed.
Please reopen a new one if needed.

closed 2024-09-02 by opeyrard

mcoracin, 2025-06-24:
Hello,

There is an example for geolocation + lorawan here:
https://github.com/Lora-net/SWL2001/tree/master/lbm_applications/3_geolocation_on_lora_edge

The corresponding main application is this file:
https://github.com/Lora-net/SWL2001/blob/master/lbm_applications/3_geolocation_on_lora_edge/main_geolocation/main_geolocation.c

By default it will configure for GNSS/WiFi scan, join a LoRaWAN network and send uplinks.
For GNSS it is using the "store and forward" feature for demonstration, but if you want to send direct uplink, you can change the line:
https://github.com/Lora-net/SWL2001/blob/9a14f6772c1d6e303bacb2d594c8063bb804b6ee/lbm_applications/3_geolocation_on_lora_edge/main_geolocation/main_geolocation.c#L291

Best regards

umamaheshrouthu, 2025-06-24:
> The corresponding main application is this file:

@mcoracin 
thank you for your reply. 
I am using nrf + lr1110. in this case, this github repo needs to be ported into zephyr.
something like this https://github.com/Lora-net/LBM_Zephyr
however, this repo donot have a sample project for gnss + lorawan or the sample corresponding to https://github.com/Lora-net/SWL2001/tree/master/lbm_applications/3_geolocation_on_lora_edge

It has only a sample project corresponding to periodic_uplink.
so, checking if such project is available