Lora-net/SWL2001 #52: Trouble joining a LoRaWAN network

EddieCarrera, 2024-04-12:
I have built a prototype PCBA board with an STM32L433 MCU interfacing with the LR1120, with the goal of evaluating the L1120. I leveraged the SemTech provided PCB design for the LR1120 layout (https://www.semtech.com/products/wireless-rf/lora-edge/lr1120dvk1tcks#documentation) . 

After learning to update the LR1120 firmware to 0x201 (with this repo https://github.com/Lora-net/SWTL001) and porting the SWL2001 driver code to work with the STM32L433, I used the “porting tests” example firmware from SWL2001 to verify that the LR1120 radio is working correctly with the STM32L433 – it passed all tests. 

From there I made some changes to the LoRaWAN network configuration “example_options.h” to get the LR1120 to connect to our network. I’m also calling the following: ASSERT_SMTC_MODEM_RC( smtc_modem_set_network_type(stack_id, false) ) to ensure the radio is configured for a private LoRaWAN network (my network is private). However, my gateway only receives a join request packet from the LR1120 intermittently (1 out of 10 tries). It’s also observed that the LR1120 never receives any downlinked “join request accepted” packet even though it’s confirmed that the gateway has sent the packet on its end. 

I have a few things I want to try to debug this network connection issue:

1) Lower the max transmit power of the LR1120 to a max of 14dBm. I thought I made the appropriate changes in the code, but the LR1120 continues to send a “join request” packet at the max 22dBm (See screenshot below). I would like to lower the max to 14dBm as a debugging step because I’ve found that a LoRa radio front-end that isn’t perfectly tuned has issues transmitting at max power – which might be the case. 

  - A follow up question, what API exists for me to choose weather to transmit with the HP vs LP path?

2) I also wanted to disable ADR on the LR1120 stack – which I couldn’t find in the code. 

[image]

I’m still in the process of trying to troubleshoot the root cause of this – as I’m unsure if it’s a firmware configuration issue, or hardware issue. As a troubleshooting step, I am no longer interfacing with the STM32L433 on the custom PCB. Instead, currently am interfacing the LR1120 (on the custom board) with the NUCLEO-L476RG (via jumper wires soldered to the SPI lines), which is what the default out-of-the-box setup for the SWTL001 LR1120 radio library example. However, the exact same behavior persists between the NUCLEO-L476RG and STM32L433 MCU. So this would indicate some sort or hardware/software issue on the LR1120 side rather than the STM32 side. 

Any further troubleshoot steps and advice would be appreciated! I have experience with STM32's stackforce LoRaWAN radio stack, so although I'm no expert, I have exposure to the config setup to make an edge device communicate via LoRaWAN.

lbm-team, 2024-04-16:
Hi EddieCarrera,
Thanks for using LBM.

Does your gateway support 64 uplink channels?
When the modem is configured with US915 region, it uses 64 channels (BW125) + 8 channels (BW500). If your gateway only supports 8 uplink channels, you will lose a lot of joinRequest, but when the join is received, the network should downlink the correct configuration based on the gateway configuration.

1) To setup HP vs LP  you have to implement your own ral_lr11xx_bsp.c as proposed in lbm_examples/radio_hal/ral_lr11xx_bsp.c

2) To disable ADR you have to use the API smtc_modem_adr_set_profile()

EddieCarrera, 2024-04-16:
Thanks for the response!

That's a very good point about the channels. That can be a contributing factor to intermittent join requests being seen by the gateway. I forgot to mention the LoRaWAN network I'm running only utilizes sub-band 1 (channels 1-8). Other LoRaWAN stacks (such at Stackforce's) has an option to enable hybrid mode which limits channel masks, and limits the application to a single sub-band (sub-band 1 in this case). An example of such an implementation is below:

```
void RegionUS915InitDefaults( InitDefaultsParams_t* params )
{
#if defined( REGION_US915 )
    Band_t bands[US915_MAX_NB_BANDS] =
    {
       US915_BAND0
    };

    switch( params->Type )
    {
        case INIT_TYPE_DEFAULTS:
        {
            if( ( params->NvmGroup1 == NULL ) || ( params->NvmGroup2 == NULL ) )
            {
                return;
            }

            RegionNvmGroup1 = (RegionNvmDataGroup1_t*) params->NvmGroup1;
            RegionNvmGroup2 = (RegionNvmDataGroup2_t*) params->NvmGroup2;

            // Initialize 8 bit channel groups index
            RegionNvmGroup1->JoinChannelGroupsCurrentIndex = 0;

            // Initialize the join trials counter
            RegionNvmGroup1->JoinTrialsCounter = 0;

            // Default bands
            memcpy1( ( uint8_t* )RegionNvmGroup1->Bands, ( uint8_t* )bands, sizeof( Band_t ) * US915_MAX_NB_BANDS );

            // Default channels
            for( uint8_t i = 0; i < US915_MAX_NB_CHANNELS - 8; i++ )
            {
                // 125 kHz channels
                RegionNvmGroup2->Channels[i].Frequency = 902300000 + i * 200000;
                RegionNvmGroup2->Channels[i].DrRange.Value = ( DR_3 << 4 ) | DR_0;
                RegionNvmGroup2->Channels[i].Band = 0;
            }
            for( uint8_t i = US915_MAX_NB_CHANNELS - 8; i < US915_MAX_NB_CHANNELS; i++ )
            {
                // 500 kHz channels
                RegionNvmGroup2->Channels[i].Frequency = 903000000 + ( i - ( US915_MAX_NB_CHANNELS - 8 ) ) * 1600000;
                RegionNvmGroup2->Channels[i].DrRange.Value = ( DR_4 << 4 ) | DR_4;
                RegionNvmGroup2->Channels[i].Band = 0;
            }

            // Default ChannelsMask
            /* ST_WORKAROUND_BEGIN: Hybrid mode */
#if ( HYBRID_ENABLED == 1 )
            RegionNvmGroup2->ChannelsDefaultMask[0] = 0x00FF;
            RegionNvmGroup2->ChannelsDefaultMask[1] = 0x0000;
            RegionNvmGroup2->ChannelsDefaultMask[2] = 0x0000;
            RegionNvmGroup2->ChannelsDefaultMask[3] = 0x0000;
            RegionNvmGroup2->ChannelsDefaultMask[4] = 0x0001;
            RegionNvmGroup2->ChannelsDefaultMask[5] = 0x0000;
#else
            RegionNvmGroup2->ChannelsDefaultMask[0] = 0xFFFF;
            RegionNvmGroup2->ChannelsDefaultMask[1] = 0xFFFF;
            RegionNvmGroup2->ChannelsDefaultMask[2] = 0xFFFF;
            RegionNvmGroup2->ChannelsDefaultMask[3] = 0xFFFF;
            RegionNvmGroup2->ChannelsDefaultMask[4] = 0x00FF;
            RegionNvmGroup2->ChannelsDefaultMask[5] = 0x0000;
#endif /* HYBRID_ENABLED == 1 */
            /* ST_WORKAROUND_END */

            // Copy channels default mask
[... 48 lines cut ...]
#define LR11XX_MIN_PWR_HP_LF -9
#define LR11XX_MAX_PWR_HP_LF 14

#define LR11XX_MIN_PWR_PA_HF -18
#define LR11XX_MAX_PWR_PA_HF 13
```

But the TRACE output when running the periodic_uplink.c example still indicates that a join request from the sensor is being sent at 22dBm (see screenshot in original post). Is there something else I should modify within ral_lr11xx_bsp. to achieve this?

2) I've previously tried exploring smtc_modem_adr_set_profile() as you suggested, but I couldn't find documentation for '@param [in] dr_custom_distribution_data Definition of the custom datarate distribution', so I wasn't sure how exactly to set it up. Would the following from smtc_modem_core/lorawan_certification/lorawan_certification.c be enough to disable it:

(lines 602-608)
```
            if( rx_buffer[1] == ( uint8_t ) LORAWAN_CERTIFICATION_ADR_OFF )
            {
                uint8_t adr_custom_data[16] = { 0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03,
                                                0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03 };
                smtc_modem_adr_set_profile( lorawan_certification_obj->stack_id, SMTC_MODEM_ADR_PROFILE_CUSTOM,
                                            adr_custom_data );
            }
```

Would you be able to point me in the right direction for more information on how to implement 'adr_custom_data'?

I have purchased and received the LR1120DVK1TCKS eval kit. I will play around with it, and keep this thread updated.

EddieCarrera, 2024-04-16:
Okay, I found the correction. As seen above, the sync word is set up for public (0x34) instead if private (0x12). It was because I did the following:

```
ASSERT_SMTC_MODEM_RC( smtc_modem_set_network_type(stack_id, false) );  
ASSERT_SMTC_MODEM_RC( smtc_modem_set_region( stack_id, MODEM_EXAMPLE_REGION ) );
```

As it turns out, smtc_modem_set_network_type() must be called **after** smtc_modem_set_region() for the sync word to stick. After making this fix, the eval-board now reliably connects to my LoRaWAN network.

EddieCarrera, 2024-04-18:
Initial implementation seems to be working, I'll go ahead and close this now

closed 2024-04-18 by EddieCarrera

zafersn, 2024-08-29:
Hi @EddieCarrera and @mluis1, I am also struggling with the similar issue rx timeout for cn470
But my problem is CONFIG_PM, it works ok without config_pm. 

One fail join attempt log:
```
INFO: Use soft secure element for cryptographic functionalities
stack_id 0
 DevNonce = 24
 JoinNonce = 0xff ff ff, NetID = 0xff ff ff
 Region = CN470_RP_1_0
255 smtc_modem.c
Modem event callback

Event received: RESET

2
INFO: smtc_modem_join_network
 Start a new join sequence now on stack 0
262 smtc_modem.c
313 main_periodical_uplink.c
318 main_periodical_uplink.c
Join is starting 

DevEUI - (8 bytes):
 66 30 31 34 63 36 39 65
JoinEUI - (8 bytes):
 00 00 00 00 00 00 00 00
DevNonce 0x19, stack_id 0
Send Payload  for stack_id = 0
1381 lr11_apps_common.c
ralstatus1:0 4194308 2
793 radio_planner.c update: 2
1381 lr11_apps_common.c
662 lr1_stack_mac_layer.c update: 3
749 lr1_stack_mac_layer.c update: 2
 TX DONE
radio_timestamp: 47934 time_off_ms: 0 jn stat:1
296 lr1mac_core.c wnds: 0 1
delay_ms 5000
rx_offset_ms:-2, rx_timeout_symb_in_ms:16, rx_window_symb: 8, board_delay_ms:2
freq rx1: 507500000
timeout prm: 3000
  RX1 LoRa at 52932 ms: freq:507500000, SF8, BW125, sync word = 0x34
  Timer will expire in 4975 ms
300 lr1mac_core.c past: 0 
1381 lr11_apps_common.c
ralstatus1:0 4195328 8
793 radio_planner.c update: 8
822 radio_planner.c update:
1381 lr11_apps_common.c
662 lr1_stack_mac_layer.c update: 5
749 lr1_stack_mac_layer.c update: 4
1RX1 Timeout for stack_id = 0

delay_ms 6000
rx_offset_ms:-2, rx_timeout_symb_in_ms:16, rx_window_symb: 8, board_delay_ms:2
freq rx2: 505300000
timeout prm: 3000
  RX2 LoRa at 53932 ms: freq:505300000, SF8, BW125, sync word = 0x34
  Timer will expire in 764 ms
1381 lr11_apps_common.c
ralstatus1:0 4195328 8
793 radio_planner.c update: 8
822 radio_planner.c update:
1381 lr11_apps_common.c
662 lr1_stack_mac_layer.c update: 5
749 lr1_stack_mac_layer.c update: 4
RX2 Timeout for stack_id = 0 Start a new join sequence in 14 seconds on stack 0
Modem event callback

Event received: JOINFAIL
```

one successful join request
```
INFO: Use soft secure element for cryptographic functionalities
stack_id 0
 DevNonce = 46
 JoinNonce = 0x21 00 00, NetID = 0x00 00 00
 Region = CN470_RP_1_0
Modem event callback

Event received: RESET

2
INFO: smtc_modem_join_network
 Start a new join sequence now on stack 0
Join is starting 

DevEUI - (8 bytes):
 66 30 31 34 63 36 39 65
JoinEUI - (8 bytes):
 00 00 00 00 00 00 00 00
DevNonce 0x2f, stack_id 0
Send Payload  for stack_id = 0
1446 lr11_apps_common.c
1446 lr11_apps_common.c
662 lr1_stack_mac_layer.c update: 3
749 lr1_stack_mac_layer.c update: 2
 TX DONE
radio_timestamp: 13408 time_off_ms: 0 jn stat:1
297 lr1mac_core.c wnds: 0 1
delay_ms 5000
rx_offset_ms:-2, rx_timeout_symb_in_ms:16, rx_window_symb: 8, board_delay_ms:2
freq rx1: 506700000
timeout prm: 3000
  RX1 LoRa at 18406 ms: freq:506700000, SF8, BW125, sync word = 0x34
  Timer will expire in 4980 ms
301 lr1mac_core.c past: 0 
1446 lr11_apps_common.c
1446 lr11_apps_common.c
662 lr1_stack_mac_layer.c update: 4
749 lr1_stack_mac_layer.c update: 4
1Receive a Valid downlink RX1 for stack_id = 0 update join procedure
 DevAddr= ba9a01
 MacRx1DataRateOffset= 0
 MacRx2DataRate= 0
 MacRx1Delay= 1
Modem event callback

Event received: JOINED

Modem is now joined 

Joined
```

here is more details.
 https://github.com/zephyrproject-rtos/zephyr/discussions/77637