Lora-net/SWL2001 #54: Activate LR-FHSS modulation on uplink messages.

BartomeuHeer, 2024-04-17:
Hello,

I have a gateway capable to demodulate LR-FHSS messages, so I wanted to test it. I know there is a component in the lr11xx drivers which is responsible for the LR-FHSS, but how could I used it to activate it on uplink messages?

I also wanted to ask if it's possible to change parameters like the Spreading Factor when sending uplink messages, if the Network Server don't use ADR.

Thank you for your time.

lbm-team, 2024-04-18:
Hi [BartomeuHeer](https://github.com/BartomeuHeer),

To use LR-FHSS with LBM stack the network server must send a NewChannelReq with the frequency and the datarate range that allow LR-FHSS.

To parameter your own ADR (spreading Factor/Bandwidth), you have to used this API
in this example, 50% of uplinks will sent at DR2 and  50% at DR3

```c
uint8_t adr_custom_data[16] = { 0x02, 0x02, 0x02, 0x02, 0x02, 0x02, 0x02, 0x02, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03 };
smtc_modem_adr_set_profile( 0, SMTC_MODEM_ADR_PROFILE_CUSTOM, adr_custom_data );
```

Best

opeyrard, 2024-09-02:
No update, so issue is closed.
Please reopen a new one if needed.

closed 2024-09-02 by opeyrard