Lora-net/SWL2001 #56: LCTT tests: MCU reboot due to lr1mac FAILSAFE EVENT and other issues

yobiiq-fostin, 2024-04-27:
Hello lbm-team, 
We encounter some issues during LCTT testing.
We will be delighted to have feedback on the issues described below.

**Environment**
- ISP4520-EU-ST module (nRF52832 + SX1261)
- LBM 4.4.0
- Region EU868

The LBM stack is ported sucessfully to the ISP4520-EU-ST module.

1. With stack v4.4.0 and RELAY TX and RX features disabled, the lr1mac FAILSAFE EVENT restarts the MCU and
then the LCTT tests fail. The stack v4.3.1 also have the same issue.
Location : in `lr1mac_core.c` file, line 196
Here are the lines in the stack that cause this failsafe event.
```C
 if( ( lr1_mac_obj->lr1mac_state != LWPSTATE_IDLE ) &&
        ( ( int32_t ) ( smtc_modem_hal_get_time_in_s( ) - lr1_mac_obj->timestamp_failsafe - FAILSAFE_DURATION ) > 0 ) )
    {
        SMTC_MODEM_HAL_PANIC( "FAILSAFE EVENT OCCUR (lr1mac_state:0x%x)\n", lr1_mac_obj->lr1mac_state );
        lr1_mac_obj->lr1mac_state = LWPSTATE_ERROR;
    }
```

2. With stack v4.4.0 and RELAY TX and RX features enabled, these LCTT tests failed: 
-  TP_A_EU868_ED_MAC_104_BV_001_A (LinkADRReq Join-Accept with DLSettings)
- TP_A_EU868_ED_MAC_104_BV_002 (Echo Play) 
Reason? : the payload max size (EU868 region) is less than 51 bytes in case of bad data rates.
Location : in `lorawan_certification.c` file, line 659 :
```C
    case LORAWAN_CERTIFICATION_ECHO_PLAY_REQ:
```
In this switch case, the following core function is called:
```C
/**************************************************/
/*   LoraWan  lr1mac_core_next_max_payload_length_get  Method     */
/**************************************************/
uint32_t lr1mac_core_next_max_payload_length_get( lr1_stack_mac_t* lr1_mac_obj )
{
#if defined( RELAY_TX )
    uint8_t max = smtc_relay_get_tx_max_payload( lr1_mac_obj );
    SMTC_MODEM_HAL_TRACE_WARNING( "Relay max payload %d\n", max );
    max = MIN( max, smtc_real_get_max_payload_size( lr1_mac_obj->real, lr1_mac_obj->tx_data_rate, UP_LINK ) );

    const uint8_t add_byte = lr1_mac_obj->tx_fopts_current_length + FHDROFFSET;

    if( max > add_byte )
    {
        max -= add_byte;
    }
    else
    {
        max = 0;
    }

    SMTC_MODEM_HAL_TRACE_WARNING( "Final max payload %d\n", max );

    return max;
#else
    return ( smtc_real_get_max_payload_size( lr1_mac_obj->real, lr1_mac_obj->tx_data_rate, UP_LINK ) -
             lr1_mac_obj->tx_fopts_current_length - FHDROFFSET );
#endif
}
```

opeyrard, 2024-05-27:
Could you please check with the latest release and let us know if this issue remains ?
Many thanks.

opeyrard, 2024-09-02:
No update, so issue is closed.
Please reopen a new one if needed.

closed 2024-09-02 by opeyrard