Lora-net/SWL2001 #57: Periodical uplink example doesn't work for me

martin-peevski, 2024-05-10:
Hello,

If here isn't the place for such support, please tell me where should I write?

I'm trying to use LBM on Nucleo-L073 with SX1261MB2xAS board.

That configuration don't require any porting as far as I know.
It only need to be built and it should work, but not.
Attached you'll find screenshots of the porting tests and periodical uplink examples terminal logs.

Porting tests result shows some minor timer error which I hope isn't a problem.

With the Periodical uplink example device is unable to join the network and resets at the 2nd attempt.-
Can you say what is the problem and what shall I do to make it work?

Best regards,
Martin Peevski

[image]
[image]

martin-peevski, 2024-05-10:
Now I measured with scope and get pulse on DIO1 only with the porting tests example.
On periodical uplink example there is no pulse on DIO1, I suppose Tx done interrupt isn't fired.
Hope this help you.

Best

opeyrard, 2024-05-27:
We didn't manage to reproduce this issue with the latest version of LBM SWL2001. Could you please try on your side ?

martin-peevski, 2024-06-10:
I already tried, that's why I'm writing here.

We need lorawan stack with size of 40kB - 45kB of MCU flash memory. I built LBM with the min. required packages and EU868 region and it takes around 68kB of MCU flash. Can you tell, will be possible to somehow fit in the required memory size?

opeyrard, 2024-06-10:
May I ask you if you could please clarify ? Is 45kB for code size stack only ? or for your full project ? Also, could you please confirm you disabled the log ?
Many thanks

martin-peevski, 2024-06-10:
I confirm - logs are disabled.
45kB for LBM only.

alexbegoon, 2024-06-23:
@martin-peevski don't you forget to call watchdog refresh?

martin-peevski, 2024-06-24:
> @martin-peevski don't you forget to call watchdog refresh?

Alex, 
No, it's reloaded in the example project.
Right now I'm on another project and since the LBM min. memory size is around 68kB of FLASH I don't think we can use it.

closed 2024-09-02 by opeyrard