Lora-net/SWL2001 #60: NUCLEO-U5A5ZJ-Q + LBM + Geolocation: Problem with  store and fwd init failed

BB-Vega, 2024-05-15:
Hello,

I've combined the examples 1_thread_x_on_stm32_u5 and 3_geolocation_on_lora_edge to create a new example for the NUCLEO-U5A5ZJ-Q Board. 

I've deleted the ThreadX stuff and used the application from 3_geolocation_on_lora_edge to get the geolocation stuff to work. 

In Addition I've modified the app_options.mk to use 
`# Add any lbm build options (ex: LBM_BUILD_OPTIONS ?= LBM_CLASS_B=yes REGION=ALL) `
`LBM_BUILD_OPTIONS ?= LBM_CLASS_B=yes REGION=ALL LBM_GEOLOCATION=yes LBM_ALMANAC=yes LBM_STORE_AND_FORWARD=yes` 

and adjusted the modem_pinout.h to use the right pins for the RADIO_LNA_CTRL.

The application gets stuck after the initialization of the geolocation services. It seems like there is a problem with the filesystem. Could you please help me with that? 

> LOG:
24-05-15 10:52:54.054: INFO: Modem Initialization
24-05-15 10:52:54.057: INFO: Use soft secure element for cryptographic functionalities
24-05-15 10:52:54.302: stack_id 0
24-05-15 10:52:54.303:  DevNonce = 382
24-05-15 10:52:54.305:  JoinNonce = 0x12 00 00, NetID = 0x13 00 00
24-05-15 10:52:54.309:  Region = EU868
24-05-15 10:52:54.312: LoRaWAN Certification is disabled on stack 0
24-05-15 10:52:54.316: lorawan_alcsync_services_init task_id 7, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.323: almanac_services_init
24-05-15 10:52:54.325: mw_gnss_scan_services_init task_id 13, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.331: mw_gnss_send_services_init task_id 14, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.337: mw_gnss_almanac_services_init task_id 15, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.343: mw_wifi_scan_services_init task_id 16, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.346: mw_wifi_send_services_init task_id 17, service_id 0, CURRENT_STACK:0 
24-05-15 10:52:54.346: Store and fwd # scanning for filesystem...
24-05-15 10:52:54.346: circularfs_scan: partially formatted partition
24-05-15 10:52:54.362: Store and fwd # no valid filesystem found, formatting.

If I comment the store and fwd stuff the application runs as espected.

Best regards,
Benedikt

opeyrard, 2025-02-17:
Hi,
We apologize as we missed your Ticket.
Are you still facing this issue ?
Many thanks,

BB-Vega, 2025-02-17:
Hi opeyrard,

Thank you very much for reaching out. Unfortunately we changed the chip & firmware in our project. 
I don't know if the error  still persists with this setup. But you could close the ticket for now if you like.

Best regards,
Benedikt

closed 2025-02-17 by opeyrard