Lora-net/SWL2001 #63: ACK on uplink and duty cycle.

BartomeuHeer, 2024-06-04:
Hello,

I wanted to know if for the periodical uplink code if we can send uplink messages with ACK. 

About the duty cycle, I have noticed that the uplink messages are hardcoded to send each 60 seconds independly of the data rate, thus not respecting the duty cycle. Is there a way to configure it so that the uplinks respect the duty cycle?

Thank you very much,

Bartomeu.

opeyrard, 2024-06-10:
Hi,
LoRa Basics Modem respects natively the duty cycle requirements. We let you experiment it by yourself.
Many thanks

BartomeuHeer, 2024-06-11:
Thank you for the response.

Also, is there a way to enable acknowledged uplink messages from the code?

opeyrard, 2024-06-25:
Hi,
You will find all the details in smtc_modem_api.h
Many thanks

closed 2025-02-17 by opeyrard