Lora-net/SWL2001 #65: Eeprom writes every join due to dev_nonce mismatch?

cmorganBE, 2024-06-06:
v4.3.0 with an internal (for now) port to esp32 (esp-idf) with the SX1262.

System isn't able to join with the gateway, its issuing joins and receiving JOINFAIL every 30-50 seconds.

Port passed the test suite.

Around these joins I'm seeing calls to smtc_modem_hal_context_store():

I've added prints of the hex digits to understand what the issue might be and a few prints around the condition in lr1mac_core_context_save() to understand what is triggering the store.

At the rate these writes are occurring it looks like a 100k write eeprom would be worn out in only 70 days, so I'm suspecting something must be wrong here...

```
I (47594) smtc_modem_hal: enable_modem_irq
I (47594) smtc_modem_hal: restore CONTEXT_LORAWAN_STACK size 36
I (47624) SWL2001: read eeprom:1040384
00 00 5C 05 FF FF FF FF FF FF 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1D 5C 7E 7A ctx.devnonce 1372 lr1_mac_obj->dev_nonce 1373
lr1mac mismatch, storing
I (47624) smtc_modem_hal: store CONTEXT_LORAWAN_STACK size 36
00 00 5D 05 FF FF FF FF FF FF 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 15 BF 1E 46 I (47774) SWL2001: write eeprom:1040384
I (47774) smtc_modem_hal: restore CONTEXT_LORAWAN_STACK size 36
I (47794) SWL2001: read eeprom:1040384
00 00 5D 05 FF FF FF FF FF FF 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 15 BF 1E 46 I (47794) smtc_modem_hal: stop_timer
I (47804) smtc_modem_hal: start_timer milliseconds 4770
I (48304) smtc_modem_hal: disable_modem_irq
```

opeyrard, 2024-06-10:
Hi,

Join dutycycle is implemented and will limit the number of join after 10 hours as described in the LoRaWAN standard.
Limitation will be : 32 joins per day in average (equivalent to 10 years of join before to reach the 100k limit of your eeprom).
If you want to reduce it at 4 joins @SF12 per day, you have to modify the join distribution to allow only SF12 for join request.

cmorganBE, 2024-06-27:
Closing until I have more info.

closed 2024-06-27 by cmorganBE