Lora-net/SWL2001 #73: LCTT test : Trigger Join Request problem

YZATOUT, 2024-07-12:
Hi,
I am using the LBM version 4.5.0 and I am testing the firmware using LCTT.

I obseved that loraMac freezes (no more packets are sent or a new Join is sent) when the tool asks the device to trigger a Join Request. However, all the other LCTT commands works fine.

When debugging I obseved that the "lorawan_join_internal_add_task( stack_id )" cannot be called as the modem task priority blocks the access in line below: 

`context->modem_task[JOIN_TASK + ( NUMBER_OF_TASKS * stack_id )].priority != TASK_FINISH`

I added the line below to leave the network when the LORAWAN_CERTIFICATION_DUT_JOIN_REQ command is received and now it works fine (Join triggered successfully and there is no problem)

`lorawan_api_join_status_clear( lorawan_certification_obj->stack_id );`

Can you check please if the fix is OK

```
    case LORAWAN_CERTIFICATION_DUT_JOIN_REQ:
        if( rx_buffer_length == LORAWAN_CERTIFICATION_DUT_JOIN_REQ_SIZE )
        {
            // leave network
           lorawan_api_join_status_clear( lorawan_certification_obj->stack_id );
            lorawan_api_set_activation_mode( ACTIVATION_MODE_OTAA, lorawan_certification_obj->stack_id );
            ret = LORAWAN_CERTIFICATION_JOIN_REQ;
        }
```

Thank you.
Best regards.

lbm-team, 2024-07-12:
Hello, thank you for using LBM
Can you check if you didn't delete the line of code inadvertently ?

https://github.com/Lora-net/SWL2001/blob/d05dad5b4df9c0e5899008e43bfce48f5498abb5/lbm_lib/smtc_modem_core/lorawan_packages/lorawan_certification/lorawan_certification.c#L560

The `lorawan_api_join_status_clear()` is present in the master branch

Best Regards

YZATOUT, 2024-07-12:
Hi,
Thank you for you quick reply.
Sorry, I posted the wrong place. 

The line added was : 
` smtc_modem_leave_network( lorawan_certification_obj[idx].stack_id );`

in the section below:

```
case LORAWAN_CERTIFICATION_JOIN_REQ: {
        if( lorawan_api_isjoined( lorawan_certification_obj[idx].stack_id ) != JOINED )
        {
            // Leave the network before starting a rejoin.
            smtc_modem_leave_network( lorawan_certification_obj[idx].stack_id );
            lorawan_join_add_task( lorawan_certification_obj[idx].stack_id );
        }
        break;
```

Note, that I observed that this API: "smtc_modem_leave_network()" was added for "DM_REJOIN" but not in the section above. I do not know if it is forgotten.

```
case DM_REJOIN: {
        uint16_t dev_nonce       = ( cmd_input->buffer[1] << 8 ) | cmd_input->buffer[0];
        uint16_t lr1mac_devnonce = lorawan_api_devnonce_get( ctx->stack_id );
        if( lr1mac_devnonce != dev_nonce )
        {
            SMTC_MODEM_HAL_TRACE_WARNING( "DM invalid devnonce to rejoin\n" );
            uint8_t dm_fields_payload[1] = { DM_INFO_SESSION };
            cloud_dm_set_info_field( ctx->stack_id, dm_fields_payload, 1, DM_INFO_NOW );
            break;
        }
        // Leave network
        smtc_modem_leave_network( ctx->stack_id );
        // Add a new join task
        lorawan_join_add_task( ctx->stack_id );
        break;
```

Thank you.

opeyrard, 2024-08-19:
Hi,
The `smtc_modem_leave_network()` command must not be used in certification to avoid killing the certification task.
Could you please test without this line added ?
Many thanks,
Best regards

YZATOUT, 2024-08-19:
Hi,
If I don't add the `smtc_modem_leave_network()`, the modem will never trigger a Join request. 

After debugging, I observed the the `lorawan_join_internal_add_task( stack_id ) `cannot be called. 
For unknown reason the `task priority` blocks the access in line below:

`context->modem_task[JOIN_TASK + ( NUMBER_OF_TASKS * stack_id )].priority != TASK_FINISH`

I did not find another workaround for this issue. 

Did you test from your side the LORAWAN_CERTIFICATION_DUT_JOIN_REQ command with LCTT?

Thank you.
Best regards.

opeyrard, 2025-02-17:
Hi,
Did you test it with the last release LBM V4.8.0 ?
Many thanks,

closed 2025-05-20 by opeyrard