Lora-net/SWL2001 #76: lbm lib, MCU_FLAGS default to nc generating error

disk91, 2024-07-29:
The MCU_FLAG default
```
# MCU - Must be provided by user
MCU_FLAGS ?= nc
```
Is source of compilation error
``` 
arm-none-eabi-gcc: warning: nc: linker input file unused because linking not done
arm-none-eabi-gcc: error: nc: linker input file not found: No such file or directory
make[4]: *** [makefiles/common.mk:535: build/sx1262/trace/ralf_sx126x.o] Error 1
make[4]: *** Waiting for unfinished jobs....
```

It would be better to have an empty value instead of `nc`

If other user have the same problem, compile w/o having to change the option.mk file with `make ... MCU_FLAG=` but the best is to set the righ mcu flags according to your target MCU.

If the purpose is to force user to setup MCU FLAGS, better replace `nc` by `please provide MCU FLAGS in options.mk`

opeyrard, 2025-02-17:
Hi,
Thank you very much for your feedback.
We will investigate it.

opeyrard, 2025-05-20:
Hi,
Could you please precise which OS you are using to compile ?
Many thanks,

closed 2026-10-07 by lbm-team