Lora-net/SWL2001 #82: Different frequncies on TX and RX1

Shubham-embed, 2024-08-15:
When trying to connect to the gateway in the Region_US_915 we are seeing that the RX1 (923.9) on the board and the TX (923.3) from the gateway are on different frequencies. We have set the board to stay in Sub-band 2 by commenting out
snapshot_bank_tx_mask++;
[image]
[image]

Achichig, 2024-08-16:
Hello, it seems that your network server is trying to send the downlink on the Rx2 window.
In US915, the Rx1 window parameters depend on the uplink that initiates the data exchange according to the following formula:
RX1 Channel Number = Transmit Channel Number modulo NbChannel, where NbChannel is the number of active receive channels.
This results in the following for the first channels:
[image]

The Rx2 window, however, will have fixed parameters regardless of the uplink: 923.3MHz/DR8 (SF12BW500).

DerekByrne, 2024-08-21:
Hey @Achichig  work with @Shubham-embed. 

It seems that the Up and expected down are off by 0.6 mHz:
In the screenshot I took 2 examples
1) Up = 903.9 (N=8)  , Down = 923.9 (N=9)
2) Up = 904.3 (N=10) , Down = 925.1 (N=3)

When tested the us_region_915.c , us_region_915.h, us_region_915_defs.h we the same as used in the code (all channels enabled) just incase there had been a change we did by mistake

The default DEFAULT_RX_FREQ_500_START_US_915 is set to 923900000 but on the graph it starts at 923300000. 

When I change it to 923300000 it seems to fix the difference in frequencies but we can still not join the network

![image_480](https://github.com/user-attach
[image]
ments/assets/e16975a1-f6d6-4ab2-8300-f8e8311cb584)

Achichig, 2024-08-21:
Hello,

Indeed, Default_RX_FREQ_500_START_US should be 923300000 and not 923900000. I just checked on the master branch (and the relay branch), and on both of these branches, this value is correctly set to 923300000 Hz.

https://github.com/Lora-net/SWL2001/blob/d05dad5b4df9c0e5899008e43bfce48f5498abb5/lbm_lib/smtc_modem_core/lr1mac/src/smtc_real/src/region_us_915_defs.h#L117C41-L117C56

opeyrard, 2024-09-02:
Thanks for your feedback.
We can close the issue.

closed 2024-09-02 by opeyrard