Lora-net/SWL2001 #85: Lora JOIN request fails & RX TIMEOUT

zafersn, 2024-08-29:
I’ve ported the Semtech LoRa Basic Modem library to Zephyr for my project. It works fine until I enable CONFIG_PM. Does anyone have any insight into what might be causing this issue? I understand that the LoRa Basic Modem library isn’t officially part of Zephyr yet, but perhaps someone has encountered a similar problem and has a solution or idea.

I can see that the join request from the chip successfully reaches the gateway, and the gateway accepts it, but the LoRa device immediately generates a fail event due to modem returns RX Timeout flags. There is no problem with dev eui and app_key. if I disable the configuration. a join request will will be ok.

Here are the gateway LoRaWAN frames:

[image]

zephyr version: v3.6-branch
sdk version: zephyr-sdk-0.16.6
MCU: STM32U575ZIQ
SHIELD: semtech_lr1121mb1xxs
Any workaround ideas? Can anyone shed some light on how to debug at least?

SUCCESFULL JOIN LOGS
```
INFO: Use soft secure element for cryptographic functionalities
[00:00:36.403,000] <dbg> lora_mac_layer: lr1_stack_mac_region_config: smtc_real_init done

[00:00:36.403,000] <dbg> lr1mac_core: lr1mac_core_init: stack_id 0

[00:00:36.403,000] <dbg> lr1mac_core: lr1mac_core_init:  DevNonce = 708

[00:00:36.403,000] <dbg> lr1mac_core: lr1mac_core_init:  JoinNonce = 0x0d 02 00, NetID = 0x00 00 00

[00:00:36.403,000] <dbg> lr1mac_core: lr1mac_core_init:  Region = CN470_RP_1_0

[00:00:36.404,000] <dbg> lora_radio_planner: rp_task_abort: RP: rp_task_abort #1

[00:00:36.404,000] <dbg> lora_mac_layer: lr1_stack_mac_region_config: smtc_real_init done

[00:00:36.404,000] <dbg> lora_certification: lorawan_certification_services_init: LoRaWAN Certification is disabled on stack 0

[00:00:36.404,000] <inf> lora_periodic: Modem event callback

[00:00:36.404,000] <inf> lora_periodic: Event received: RESET

2
[00:00:36.404,000] <dbg> lora_mac_layer: lr1_stack_mac_region_config: smtc_real_init done

INFO: smtc_modem_join_network
 Start a new join sequence now on stack 0
[00:00:36.404,000] <dbg> lora_module: lora_callback_handler: SMTC_MODEM_EVENT_RESET
[00:00:36.404,000] <inf> lora_periodic: Join is starting 

[00:00:36.405,000] <dbg> lora_mac_layer: lr1_stack_mac_region_config: smtc_real_init done

[00:00:36.405,000] <dbg> lora_radio_planner: rp_task_enqueue: RP: Task #2 enqueue with #7 priority

[00:00:36.405,000] <dbg> lora_radio_planner: rp_task_arbiter:  RP: Arbiter has been called by rp_task_enqueue and priority-task #2, timer hook #0, delay 2448, now 36405
                        
[00:00:36.405,000] <dbg> lora_radio_planner: rp_task_arbiter:  RP: High priority task is in the future

[00:00:36.413,000] <inf> modem_hl7812: RTC string: '"00/00/00,00:00:00-00"'
[00:00:38.846,000] <dbg> lora_radio_planner: rp_task_arbiter:  RP: Arbiter has been called by rp_timer_irq and priority-task #2, timer hook #2, delay 7, now 38846
                  
[00:00:38.846,000] <dbg> lora_radio_planner: rp_task_launch_current:  RP: Launch task #2 and start radio state 2, type 12

[00:00:38.846,000] <dbg> lora_radio_planner: rp_task_print: 
                                                            RP- INFO - Radio task #2  running - Timer task #2 running  - Hook ID #2 -
[00:00:38.846,000] <dbg> lora_radio_planner: rp_task_print:  TASK_ERROR 
[00:00:38.846,000] <dbg> lora_radio_planner: rp_task_print:  - start time @38853 - priority #7

[00:00:38.855,000] <dbg> lora_radio_planner: rp_callback:  RP: INFO - Radio IRQ received for hook #2

DevEUI - (8 bytes):
 66 30 31 34 63 33 65 39
JoinEUI - (8 bytes):
 00 00 00 00 00 00 00 00
[00:00:38.855,000] <dbg> lora_mac_layer: lr1_stack_mac_join_request_build: DevNonce 0x2c5, stack_id 0
[... 260 lines cut ...]

[00:00:46.820,000] <dbg> lora_radio_planner: rp_callback:  RP: INFO - Radio IRQ received for hook #1

[00:00:46.820,000] <dbg> lora_app_common: radio_on_dio_irq: 1446 lr11_apps_common.c

[00:00:46.821,000] <dbg> lora_mac_layer: lr1_stack_mac_rp_callback: 662 lr1_stack_mac_layer.c update: 5

[00:00:46.821,000] <dbg> lora_mac_layer: lr1_stack_mac_rp_callback: DR0 Fine tune correction (ms) = 0, error fine tune (ms) = -5, lr1_mac->rx_offset_ms = -2

[00:00:46.821,000] <dbg> lora_mac_layer: lr1_stack_mac_rp_callback: 749 lr1_stack_mac_layer.c update: 4

[00:00:46.821,000] <dbg> lora_radio_planner: rp_task_arbiter:  RP: No more active tasks

[00:00:46.821,000] <dbg> lr1mac_core: lr1mac_core_process: RX2 Timeout for stack_id = 0
 Start a new join sequence in 14 seconds on stack 0
[00:00:46.821,000] <inf> lora_periodic: Modem event callback

[00:00:46.821,000] <inf> lora_periodic: Event received: JOINFAIL

[00:00:46.821,000] <dbg> lora_module: lora_callback_handler: SMTC_MODEM_EVENT_JOINFAIL
```

I created the same issue here too:

https://www.thethingsnetwork.org/forum/t/lora-join-request-fails-rx-timeout/69045/8

opeyrard, 2024-09-02:
Hi,
A porting of LBM on Zephyr OS is now available as a feature branch [here](https://github.com/Lora-net/LBM_Zephyr).
Many thanks,
Best regards,

jakeru, 2024-09-03:
Hi,
I have been playing a bit with the LBM Zephyr port from Semtech on a RAK4631 module. My node is able to join also when I enable Zephyr Power Management for the kernel (`CONFIG_PM=y`) as well as for devices (`CONFIG_PM_DEVICE=y`). 
Note that the `CONFIG_PM_DEVICE` option is actually not implemented for the LoRa radio drivers. The sx1262 is put into sleep mode by the driver regardless of that option.

Salamandar, 2024-09-03:
Hi @jakeru , indeed, the device is put in sleep mode by the LBM stack and not by the zephyr-side driver. This can be open to discussion, but IMHO the current behaviour is "sane" enough.