Lora-net/SWL2001 #89: LR1110: how to erase almanac data

tve, 2024-09-11:
I'd like to perform a number of tests around GNSS cold-start and I wonder how I can erase the almanac data in an LR1110 (firmware v4.1). Alternatively, is there an old almanac I can load? Or perhaps a way to construct an empty almanac?

mcoracin, 2024-09-16:
Hello,
In order to flash a particular almanac, you can use the following tool provided with this project:
https://github.com/Lora-net/SWL2001/tree/master/lbm_applications/3_geolocation_on_lora_edge/main_full_almanac_update

The readme indicates how to get an almanac image from loracloud.com and compile the tool with it to flash it.

It is indicated how to flash the most recent almanac image, but you can add the -g option to the get_full_almanac.py script with a GPS time to get an older almanac image.

It is also possible to reprogram the complete lr1110 firmware image (this will reset the almanac too) with the LR11xx updater tool:
https://github.com/Lora-net/SWTL001

Best regards

tve, 2024-09-16:
Thanks! I will try that.