Lora-net/SWL2001 #90: lr11xx_gnss_doppler_solver_result_s info?

tve, 2024-09-12:
I'm looking for info on `lr11xx_gnss_doppler_solver_result_s` https://github.com/Lora-net/SWL2001/blob/master/lbm_lib/smtc_modem_core/radio_drivers/lr11xx_driver/src/lr11xx_gnss_types.h#L436C16-L448

Specifically:
- what are the units for one_shot_accuracy / filtered_accuracy ?
- what does the filtering consist of?

mcoracin, 2024-09-16:
Hello,
The units for one_shot_accuracy / filtered_accuracy is kilometers.

Concerning the filtering, when there is scan with enough number of doppler, the algorithm will first use all valid (almanac not too old) dopplers to do a doppler solving to get a one-shot location estimate (Latitude Longitude accuracy and Xtal). Then it will check the time elapsed from the last doppler based location update. If this time is too long (currently set as 4 hours), the latest one-shot doppler solver results will be directly used as filtered output, otherwise, a low pass filter will be used to filter the latest one-shot result with the old results.

Best regards

tve, 2024-09-16:
Thanks you for the info! The 4 hour time limit is good to know as I was going to run tests with a 4 hour delay between fixes and thus probably would just miss the window. I better run tests at 3.5 hours and also at a longer interval...

closed 2024-11-12 by opeyrard