Lora-net/SWL2001 #91: Radio pin interrupt sometime not registered

RobTWTG, 2024-09-18:
When further investigating this old issue: https://github.com/Lora-net/SWL2001/issues/77
I observed that sometimes a RX_DONE task is not finished properly, and the lbm will still panic like this:

`[ERR ][LHAL]: function: rp_callback, line: 443
[ERR ][LHAL]: RP_FAILSAFE - #2`

I noticed that in the rp_callback function the 1st thing that gets done is `smtc_modem_hal_disable_modem_irq( );` disabling the timer and pin interrupts.

With overlapping tasks (like managing ping slots for class B) It looks like there is a chance that radio pin interrupts can be missed. Because `rp_callback` is called periodically.

I ran a simple test removing all functionality from `smtc_modem_hal_disable_modem_irq` so not disabling the interrupts anymore, and the panics disappeared.

Hopefully this can get looked into.

Many thanks,
Best regards

closed 2025-02-17 by opeyrard

mbp-flonidan, 2025-02-24:
@opeyrard was this issue closed because of the recent (version 4.8.0) change to `rp_callback` ?

I have an issue with `smtc_modem_hal_disable_modem_irq` and `smtc_modem_hal_enable_modem_irq` not being called at all after upgrade from 4.5.0 to 4.8.0 and I just want to know if I should raise a new issue?
As an example. In version 4.5.0 `smtc_modem_hal_enable_modem_irq` would get called prior to transmission, which is not the case anymore causing no TX irq to be triggered.
The only references I can find to `smtc_modem_hal_disable_modem_irq` and `smtc_modem_hal_enable_modem_irq` are now in fifo_ctrl.c.

MartinBP, 2025-02-24:
@opeyrard was this issue closed because of the recent (version 4.8.0) change to `rp_callback` ?

I have an issue with `smtc_modem_hal_disable_modem_irq` and `smtc_modem_hal_enable_modem_irq` not being called at all after upgrade from 4.5.0 to 4.8.0 and I just want to know if I should raise a new issue?
As an example. In version 4.5.0 `smtc_modem_hal_enable_modem_irq` would get called prior to transmission, which is not the case anymore causing no TX irq to be triggered.
The only references I can find to `smtc_modem_hal_disable_modem_irq` and `smtc_modem_hal_enable_modem_irq` are now in fifo_ctrl.c.