Lora-net/SWL2001 #92: Relay ACK MIC incorrectly calculated

StevenCellist, 2024-10-09:
Hi team,

In my opinion, the MIC of the ACK message is not correctly calculated. According to TS011-1.0.0 section 6.2, the CMAC is calculated over block B_0, AckUplinkEnc and WOR, where "WOR contains the data received **in** the WOR Relay Class A Uplink".
I read this as follows: the Frequency and Datarate fields in the WOR block should the Frequency and Datarate _as received in the WOR frame_.
However, in LBM, the MIC is calculated using the Frequency and Datarate of the ACK message:

https://github.com/Lora-net/SWL2001/blob/0ad056afad1c2102b8bcd121e87cc060d46d98c0/lbm_lib/smtc_modem_core/lr1mac/src/relay/relay_tx/relay_tx.c#L596-L609

https://github.com/Lora-net/SWL2001/blob/0ad056afad1c2102b8bcd121e87cc060d46d98c0/lbm_lib/smtc_modem_core/lr1mac/src/relay/common/wake_on_radio.c#L284-L287

I think that these fields should refer to the WOR payload's Freq/Dr instead. Please let me know your thoughts!

mluis1, 2024-10-10:
Duplicate of #62

StevenCellist, 2024-10-10:
Ah, sorry, the name of that issue somehow didn't strike my eye as being a duplicate. Thanks! I'll close this one.

closed 2024-10-10 by StevenCellist