Lora-net/SWL2001 #94: Assistance Needed for LoRa 2.4GHz Configuration and Join Request Issues

sluissantos, 2024-10-18:
Hello,

I am working on a project using **LoRa 2.4GHz** and need some assistance with configuration and performance.

Here are the details of my setup:

- **Development Environment:** SES 8.18, nRF5 SDK Version 17.1, nRF52832.
- **Transceiver:** E28-2G4M12S (SX1280).
- **Gateway:** MikroTik wAP LR2 kit.

I am adapting the example from `application_example/2_porting_nrf_52840` to work with the **nRF52832**. For testing, I am using the file `main_periodical_uplink.c` and here is the output I get:

```
<info> app: INFO: Modem Initialization
<info> app: INFO: Use soft secure element for cryptographic functionalities
<info> app: stack_id 0
<info> app: DevNonce = 0
<info> app: JoinNonce = 0xff ff ff, NetID = 0xff ff ff
<info> app: Region = WW2G4
<info> app: LoRaWAN Certification is disabled on stack 0
<info> app: Modem event callback
<info> app: INFO: Event received: RESET
<info> app: INFO: smtc_modem_join_network
<info> app: Start a new join sequence in 3 seconds on stack 0
<info> app: INFO: Periodical uplink example is starting 
<info> app: STMC_MODEM_STATUS = 0
<info> app: INFO: No network joined yet or unknown status
<info> app: STMC_MODEM_STATUS = 0
<info> app: INFO: No network joined yet or unknown status
<info> app: DevEUI - (8 bytes): 00 00 00 00 00 00 00 00
<info> app: JoinEUI - (8 bytes): 00 00 00 00 00 00 00 00
<info> app: DevNonce 0x1, stack_id 0
<info> app: INFO: SMTC_MODEM_STATUS_JOINING
<info> app: *************************************
<info> app: Send Payload  for stack_id = 0
<info> app: *************************************
<info> app: Tx LoRa at 9014 ms: freq:2479000000, SF10, BW800, len 23 bytes 8 dBm, fcnt_up 0, toa = 75
<info> app: INFO: SMTC_MODEM_STATUS_JOINING
<info> app: INFO: SMTC_MODEM_STATUS_JOINING
```

While monitoring the traffic on my gateway, I rarely receive the join request message. When I do receive it, the **RSSI** is usually below -100 dBm, which indicates poor signal strength.

Here are my gateway configurations:

- General settings:
  [image]
  
- Channels (all configured to **SF12**):
  [image]

- Join messsage (very rarely) receiveed:
  [image]

I am trying to ensure that I can successfully send messages.

**My questions:**
1. How can I configure the radio to always use Spreading Factor 12 (SF12) (along with all other radio parameters), even for the join request message?
2. The RSSI is consistently low. What could be the possible cause for this poor signal quality?

Thank you for your help!

sluissantos, 2024-11-21:
The issue is that my tasks were not running.

Now, I can send uplinks, but I get the following error in the ChirpStack Gateway Bridge logs:

nov 21 10:32:51 santos chirpstack-gateway-bridge[35649]: {"addr":{"IP":"192.168.88.1","Port":57128,"Zone":""},"data_base64":"AgGpAJNX70hTWDU5eyJyeHBrIjpbeyJjaGFuIjoxLCJjb2RyIjoiT0ZGIiwiZGF0YSI6IiIsImRhdHIiOiJTRjEyQlc4MTIiLCJmb2ZmIjotMTkwNTQ2LCJmcmVxIjoyNDc5LjAwMDAwMCwibHNuciI6LTI0LjAwMDAwMCwibW9kdSI6IkxPUkEiLCJyc3NpIjotMTA0LCJzaXplIjowLCJzdGF0IjoxLCJ0bXN0Ijo0NDUwNjc1NDR9XX0=","error":"get uplink frames error: backend/semtechudp/packets: get uplink frame error: backend/semtechudp:packets: invalid CodR: OFF","level":"error","msg":"backend/semtechudp: could not handle packet","time":"2024-11-21T10:32:51-03:00"}

I’ve tried different Coding Rates, but the error persists.

opeyrard, 2025-02-17:
Hi,
The Coding Rate (CR3) used seems wrong. This should be Coding Rate 4/8, which corresponds to CR7 in the RAL.
For more details, please refer to https://github.com/Lora-net/SWL2001/blob/master/lbm_lib/smtc_modem_core/lr1mac/src/smtc_real/src/region_ww2g4_defs.h

Could you please try and let us know ?
Many thanks,

closed 2026-10-07 by lbm-team