Lora-net/SWL2001 #96: Join issue at startup - STM32WB55 with LR1110

mikymike006, 2024-11-30:
Hello team,

I'm stuck with the join procedure.
My design is a copy of the Lora tracker with STM32WB55 and LR1110. I first managed to make it work on one board and join the network. The four other boards I have don't work. So I used the SWSD004 and compiled it with STM32Cube. It was painful.
The code work on the first board again but not on the others. I added the flag MODEM_HAL_DBG_TRACE to get more info.
By doing this I noticed that the LR1110 talk with the STM32 but fail when we try to join network on the other boards :
`[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is mute`

I tried again the same code on the working board, it worked the first time... but not since then.
Now I got this : 
`LR11XX FW : 0x0401
[1B][0;32mINFO: Modem Initialization
[1B][0m[1B][0;32mINFO: Use lr11xx crypto engine for cryptographic functionalities
[1B][0m[1B][0;33mWARN: Use lr11xx preprovisioned EUIs and keys
[1B][0mstack_id 0
 DevNonce = 2
 JoinNonce = 0xff ff ff, NetID = 0xff ff ff
 Region = EU868
[1B][0;32mINFO: LoRaWAN Certification is disabled on stack 0 
[1B][0malmanac_services_init
mw_gnss_scan_services_init task_id 8, service_id 0, CURRENT_STACK:0 
mw_gnss_send_services_init task_id 9, service_id 0, CURRENT_STACK:0 
mw_gnss_almanac_services_init task_id 10, service_id 0, CURRENT_STACK:0 
mw_wifi_scan_services_init task_id 11, service_id 0, CURRENT_STACK:0 
mw_wifi_send_services_init task_id 12, service_id 0, CURRENT_STACK:0 
Store and fwd # scanning for filesystem...
Store and fwd # found existing filesystem, usage: 0/1450
CIRCULARFS read: {0,0} cursor: {0,0} write: {0,0}
[0000] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0001] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0002] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0003] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0004] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0005] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0006] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0007] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0008] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0009] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0010] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0011] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0012] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0013] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0014] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0015] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0016] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0017] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0018] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0019] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0020] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0021] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0022] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0023] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0024] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0025] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0026] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0027] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0028] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[0029] [v=0x00000001] [FREE      ] EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE
[1B][0;32mINFO: ###### ===== BASICS MODEM RESET EVENT ==== ######
[1B][0m[1B][0;32mINFO: ###### ===== LoRa Basics Modem reset ==== ######

[1B][0m[1B][0;31mERROR: In ../apps/tracker_application/main_tracker_application_autonomous.c - main (line 318): SMTC_MODEM_RC_FAIL
[1B][0m[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is suspend
[1B][0m[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is suspend
[1B][0m[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is suspend
[1B][0m[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is suspend
[1B][0m[1B][0;32mINFO: smtc_modem_join_network
[1B][0m[1B][0;33mWARN: smtc_modem_join_network call but the device is suspend
[1B][0m[1B][0;31mERROR: In ../apps/tracker_application/main_tracker_application_autonomous.c - main (line 321): SMTC_MODEM_RC_FAIL
[1B][0m`

How can the LR1110 be in suspend mode ? I don't understand what this means.
Sometimes it also says that it's joined but it's wrong.

Could you help me please ?
Best regards,
Michael

lbm-team, 2024-11-30:
Hello, thank you for using LBM
Can you check if you are followy the prerequisites :
https://github.com/Lora-net/SWL2001?tab=readme-ov-file#prerequisites

It seems the code is not builded with the correct version of arm gcc
Best

mikymike006, 2024-12-01:
Hello LBM team,
Thank you for this fast reply.
You were right, I installed the recommended GNU arm gcc toolchain and my first board is working again. But not the others. They fail when we start to do something network related.
Here is the log from the working board, followed by the log from a non working board : 

[Log 88CA - gcc10.txt](https://github.com/user-attachments/files/17969750/Log.88CA.-.gcc10.txt)
[Log 88F2 - gcc10.txt](https://github.com/user-attachments/files/17969756/Log.88F2.-.gcc10.txt)

I see different values for DevNonce, if that's relevant.
What would you recommend to find what's going on ?

Best regards,
Michael

opeyrard, 2025-02-17:
Hi,
Thank you very much for your message.
Are the 2 Hardware 100% the same, specifically regarding the TCXO vs XTAL (32MHz) ?
Many thanks,

closed 2026-10-07 by lbm-team