Lora-net/SWL2001 #99: LCTT TP_RELAY_EU868_ED_MAC_104_BV_003 Fails on tag v4.6.0

sca-l, 2025-01-10:
Hi all,
I'm using the Semtech source code (https://github.com/Lora-net/SWL2001) to build the ED with relay functionalities (SWL2001/master tag 4.5.0) and the Relay node ((SWL2001/features/relay tag 4.6.0).
I'm using the STM32WL5 platform for both devices.
All seems work properly and many class A tests on LCTT are executed with PASS result.
First test related to Relay TP_RELAY_EU868_ED_MAC_104_BV_002 is PASS. 
Instead the test TP_RELAY_EU868_ED_MAC_104_BV_003 works properly up to the Step 05 - RelayConfReq that fails sistematically on LCTT. 
The issue occurs on the step that should re-enable the relay functionality. Seems that the RelayConfReq command is correctly parsed, but the relay service is not correctly re-enabled.
In particular, the relay_stop function must be called before to relay_start function for reset the state and abort the tasks.
This causes a partial restart of relay service without CAD schedule. So the ED is not able to communicate again through the Relay. 

The source code part that should allow to re-enable correctly the relay feature is currently commented.

It is in file relay_rx.c lines 582-588:

    // Memorise if relay is running to restart it (or not) at the end
    // bool relay_is_running = false;
    // if( relay_info.state > CAD_STATE_INIT )
    // {
    //     relay_stop( false );
    //     relay_is_running = true;
    // }
	
Enabling this part this Step 05 PASS.

I also verified the source code in the new tag 4.8.0 and it is commented yet.

Is there any reason why this part is commented?

Best regard,
Laura Scalzo

opeyrard, 2025-02-17:
Hi,
Thank you very much for your feedback.
On our side, we passed this test with success using LBM V4.8.0
Did you also try passing it with this last release ?
Many thanks

sca-l, 2025-04-28:
Hi,
I tried with the latest LBM v4.8.0 and test PASS. So I think it is something fixed in some way in this release.
Thank you, regards,
Laura Scalzo

closed 2025-04-28 by lbm-team