Lora-net/SWL2001 #104: Compile error with IAR - argument of type "smtc_modem_class_b_ping_slot_periodicity_t *" is incompatible with parameter of type "uint8_t *"

mbp-flonidan, 2025-03-11:
In this function `smtc_modem_multicast_class_b_get_session_status` located in smtc_modem.c there is a call to `lorawan_api_multicast_b_get_session_status` which takes an uint8_t pointer argument, but a pointer to `smtc_modem_class_b_ping_slot_periodicity_t` is provided.

IAR complains about this with the following error.
Error[Pe167]: argument of type "smtc_modem_class_b_ping_slot_periodicity_t *" is incompatible with parameter of type "uint8_t *"

Code copied in below for reference.
```
smtc_modem_return_code_t smtc_modem_multicast_class_b_get_session_status(
    uint8_t stack_id, smtc_modem_mc_grp_id_t mc_grp_id, bool* is_session_started, bool* is_session_waiting_for_beacon,
    uint32_t* freq, uint8_t* dr, smtc_modem_class_b_ping_slot_periodicity_t* ping_slot_periodicity )

{
#if defined( SMTC_MULTICAST )
    RETURN_BUSY_IF_TEST_MODE( );
    RETURN_INVALID_IF_NULL( is_session_started );
    RETURN_INVALID_IF_NULL( is_session_waiting_for_beacon );
    RETURN_INVALID_IF_NULL( freq );
    RETURN_INVALID_IF_NULL( dr );
    RETURN_INVALID_IF_NULL( ping_slot_periodicity );

    smtc_modem_return_code_t modem_rc;
    lorawan_multicast_rc_t   rc = lorawan_api_multicast_b_get_session_status(
        mc_grp_id, is_session_started, is_session_waiting_for_beacon, freq, dr, ping_slot_periodicity, stack_id );
```

lbm-team, 2025-04-15:
Thank you for your report, we will fix that
best

mbp-flonidan, 2025-04-15:
Thanks for your feedback @lbm-team .

Just a small note. I see that you do a cast from the "enum pointer" to uint8_t* in other similar functions.
One example of this is sx126x_get_pkt_type.

Someone has raised another issue related to this function. See https://github.com/Lora-net/SWL2001/issues/110

Another solution to that issue and the one reported in this ticket can also be resolved by creating a variable on the stack of type uint8_t as follows.

```
uint8_t ping_slot_periodicity_copy;
lorawan_multicast_rc_t   rc = lorawan_api_multicast_b_get_session_status(
        mc_grp_id, is_session_started, is_session_waiting_for_beacon, freq, dr, &ping_slot_periodicity_copy, stack_id );
*ping_slot_periodicity = ( smtc_modem_class_b_ping_slot_periodicity_t ) ping_slot_periodicity_copy;
```

I'm sure you will find a solution.
Thanks.

closed 2025-10-15 by lbm-team