Lora-net/SWL2001 #117: Make this repo a Libray?

dala318, 2025-05-09:
Would be great if this repo also could serve as a library that you could include in other projects.
Ex: Platform.io
* add a libary.json in root (the easy part)
* rename lbm_lib/ to src/ (likely a trickier task)

The issue being now that even if I in my fork of the repo add a library.json the platform.io environment tries to compile the entire repo, including the examples etc. (which fails to compile). If the repo on the other hand had a src/ and include/ folder in root it would inform the environment to only compile those specific files in those folders.

opeyrard, 2025-05-20:
Hi,
Thanks for your feedback.
We will probably take this into consideration for a future release which will plan to release such example.
Many thanks

closed 2025-05-20 by opeyrard

dala318, 2025-05-21:
Thanks for answering. Is there a roadmap somewhere in which such users can track planning an progress on new features? And would you consider creating a feature-branch for it?
In my referenced draft /pull/119 above I have proposed som necessary changes, but could re-direct it to a feature-branch for more to contribute.

dala318, 2025-05-28:
What would improve possibility to use as external library in either Platform.io or Arduino would be to have the `lbm_lib` folder split into `src` and `include` folders in root. Then you could avoid a lot of the filters and build-flags (don't think Arduino even have support for such)