Lora-net/SWL2001 #122: lr1mac task aborted by the radioplanner message clarification

djObsidian, 2025-06-02:
Greetings. I'm developing an application using SWL2001 and at the moment I can't figure out one debug message:
`lr1mac task aborted by the radioplanner`
What exactly does it mean, in what cases is it displayed and how can I fix it?
I assume that it has something to do with timings, namely the time of entering power saving modes. While I was testing with SLEEP mode - I never got this message, but when I changed it to STOP2 (with reinitialization of peripherals) - it sometimes occurs. 
Although it has no visible effect on stack operation - all packets reach the server, I would like to know where to look for the cause and how to fix it.

opeyrard, 2025-06-03:
Hi,
Would it be possible for you to provide the detailed logs of the failure ?
Many thanks,
Best regards

djObsidian, 2025-07-02:
It looks like RX2 has been missed in my case. This was probably due to prints taking more time than expected, after I started using UART with DMA - the issue is gone. 
But this is counter-intuitive, calling `smtc_modem_run_engine( )` after RX1 returned a value clearly greater than 1000ms, even though RX2 should happen 1000ms later. 
My code should be safe from timing issues by not calling any heavy processing if `smtc_modem_run_engine( )` should be called in less than 5 seconds, but it turns out this is not the most reliable way to know that we are not expected to need to call a modem function and that we can run blocking code safely.
Any recommendations on how to deal with this other than moving the application to RTOS?

lbm-team, 2025-07-17:
Hi djObsidian,

An aborted task doesn't necessarily indicate a failure. In your case, it appears that Class C is running on radio planner task #7, while Class A is using task #2.

Since Class C has the lowest priority, it can be preempted by any higher-priority task. So, if Class A needs to send an uplink or open its receive windows, the radio planner will abort the ongoing Class C operation to execute the higher-priority Class A task.

Best

djObsidian, 2025-07-17:
Thanks for the quick response. This is an issue in this case as we are currently testing FUOTA and getting packet loss on the endpoint side. Now we are waiting for the debug kit from your example (Nucleo L476RG+SX1262MB2CAS), so we cannot test FUOTA in your `main_periodical_uplink.c` example yet.
Nevertheless, our target platform is quite similar: STM32L431RC+sx1262. The HAL implementation is written based on your `smtc_hal_l4`, the program logic is the same as `main_periodical_uplink.c`. This code passes `main_porting_tests.c` with one warning, `porting_test_timer_irq` is 2 ms longer than expected, which is apparently due to STM32 HAL. 
I attach the full log from the end node of the FUOTA update process. We would be grateful if you could help analyze this log and help us understand what is going wrong at the top level.

We are using Chirpstack Version: v4.12.1 as our Network server. There is no packet loss on the gateway side, I monitored the air with SDR, all packets were being broadcast

[teraterm6-margin18.log](https://github.com/user-attachments/files/21290650/teraterm6-margin18.log)

lbm-team, 2025-07-17:
As previously mentioned, Class A has a higher priority than Class C. When Class A transmits an uplink or opens its Rx1/Rx2 reception windows, Class C reception (RxC) is temporarily disabled, which may result in the loss of FUOTA packets.

This behavior can be observed by monitoring the radio's power consumption and analyzing the RF link's power level using a power level detector.

lbm-team, 2025-09-08:
Yes, avoiding uplink transmissions in Class A can improve the success rate of FUOTA by reducing the risk of on-air collisions.

closed 2026-10-07 by lbm-team