Lora-net/SWL2001 #128: Periodical Uplink and LCTT Certif example spamming uplinks

ThorSporringMii, 2025-07-18:
I cannot for the life of me get the LBM relay RX example to work properly. The following are my build parameters:

MODEM_APP=PERIODICAL_UPLINK (same issue with LCTT_CERTIF)
ALLOW_RELAY_TX=no
ALLOW_RELAY_RX=yes
APP_TRACE=yes

I am using a SX1261MB2xAS mbed shield on the NUCLEO-L476RG as recommended

After joining the device to ChirpStack, it runs into this infinite loop where it transmits uplinks every 5 seconds, even though the PERIODICAL_UPLINK_DELAY_S is set to 60

What am i missing here? I would expect the device to start the CAD cycle and transmit a service uplink every 60 seconds.

Here is the UART output:

``` pwsh
INFO: Modem Initialization
10:20:30:644 -> INFO: Use soft secure element for cryptographic functionalities
10:20:30:661 -> stack_id 0
10:20:30:662 ->  DevNonce = 5542
10:20:30:664 ->  JoinNonce = 0xff 00 00, NetID = 0x00 00 00
10:20:30:667 ->  Region = EU868
10:20:30:669 -> WARN: ***********************************************
10:20:30:674 -> WARN:   LoRaWAN CERTIFICATION is ENABLED on stack 0
10:20:30:680 -> WARN: ***********************************************
10:20:30:687 ->  lorawan_relay_rx_services_init task_id 7, service_id 0, CURRENT_STACK:0 
10:20:30:693 -> Modem event callback
10:20:30:695 -> INFO: Event received: RESET
10:20:30:699 -> INFO: Certification example is starting
10:20:30:704 -> INFO: Push button to enable/disable certification
10:20:30:709 ->  Start a new join sequence in 4 seconds on stack 0
10:20:37:158 -> DevEUI - (8 bytes):
10:20:37:160 ->  FF AA EE BB EE EE EE EE
10:20:37:163 -> JoinEUI - (8 bytes):
10:20:37:164 ->  00 00 00 00 00 00 00 01
10:20:37:167 -> DevNonce 0x15a7, stack_id 0
10:20:37:170 -> 
10:20:37:170 ->   *************************************
10:20:37:173 ->   * Send Payload  for stack_id = 0
10:20:37:176 ->   *************************************
10:20:37:180 ->   Tx  LoRa at 6516 ms: freq:868500000, SF10, BW125, len 23 bytes 14 dBm, fcnt_up 0, toa = 371
10:20:37:569 -> 
10:20:37:569 ->   *************************************
10:20:37:572 ->   *  TX DONE
10:20:37:573 ->   *************************************
10:20:37:603 -> 
10:20:37:603 ->   Open RX1 for Hook Id = 2  RX1 LoRa at 11933 ms: freq:868500000, SF10, BW125, sync word = 0x34
10:20:37:612 ->   Timer will expire in 4975 ms
10:20:43:029 -> 
10:20:43:029 ->   *************************************
10:20:43:032 ->   * Receive a Valid downlink RX1 for stack_id = 0
10:20:43:037 ->   *************************************
10:20:43:040 ->  update join procedure
10:20:43:042 -> MacTxFrequency [0] = 867100000 
10:20:43:045 -> MacDataRateChannel [0] =  1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 
10:20:43:051 -> MacChannelIndexEnabled [0] = 1 
10:20:43:053 -> MacTxFrequency [1] = 867300000 
10:20:43:056 -> MacDataRateChannel [1] =  1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 
10:20:43:064 -> MacChannelIndexEnabled [1] = 1 
10:20:43:065 -> MacTxFrequency [2] = 867500000 
10:20:43:067 -> MacDataRateChannel [2] =  1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 
10:20:43:073 -> MacChannelIndexEnabled [2] = 1 
10:20:43:076 -> MacTxFrequency [3] = 867700000 
10:20:43:078 -> MacDataRateChannel [3] =  1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 
10:20:43:084 -> MacChannelIndexEnabled [3] = 1 
10:20:43:087 -> MacTxFrequency [4] = 867900000 
10:20:43:090 -> MacDataRateChannel [4] =  1 1 1 1 1 1 0 0 0 0 0 0 0 0 0 0 
10:20:43:098 -> MacChannelIndexEnabled [4] = 1 
10:20:43:099 ->  DevAddr= 591f62
10:20:43:099 ->  MacRx1DataRateOffset= 0
[... 193 lines cut ...]
10:21:09:692 ->   *************************************
10:21:09:696 ->   Tx  LoRa at 39032 ms: freq:867100000, SF9, BW125, len 17 bytes 14 dBm, fcnt_up 7, toa = 165
10:21:09:883 -> 
10:21:09:883 ->   *************************************
10:21:09:886 ->   *  TX DONE
10:21:09:887 ->   *************************************
10:21:09:895 -> 
10:21:09:895 ->   Open RX1 for Hook Id = 2  RX1 LoRa at 40241 ms: freq:867100000, SF9, BW125, sync word = 0x34
10:21:09:903 ->   Timer will expire in 992 ms
10:21:10:640 -> WARN:  RP: Aborted task with hook #7 - not a priority task
10:21:10:646 ->  WARN: Relay Task aborted
10:21:10:649 -> 
10:21:10:911 ->   *************************************
10:21:10:915 ->   * RX1 Timeout for stack_id = 0
10:21:10:921 ->   *************************************
10:21:10:925 -> 
10:21:10:925 ->   Open RX2 for Hook Id = 2  RX2 LoRa at 41268 ms: freq:869525000, SF12, BW125, sync word = 0x34
10:21:10:934 ->   Timer will expire in 989 ms
10:21:11:641 -> WARN:  RP: Aborted task with hook #7 - not a priority task
10:21:11:647 ->  WARN: Relay Task aborted
10:21:11:650 -> 
10:21:12:111 ->   *************************************
10:21:12:114 ->   * RX2 Timeout for stack_id = 0
10:21:12:117 ->   *************************************
```

lbm-team, 2025-07-18:
Hi ThorSporringMii,

It seems you ran the certification mode previously, the certification state is saved in flash.

This log show that the Certification is enabled at boot:

```
10:20:30:669 -> WARN: ***********************************************
10:20:30:674 -> WARN:   LoRaWAN CERTIFICATION is ENABLED on stack 0
10:20:30:680 -> WARN: ***********************************************
```

You have to disable to stop certification uplinks every 5s.

Best Regards

ThorSporringMii, 2025-07-21:
Hello lbm-team.

Since I'm using the main_periodical_uplink.c example, there isn't an option in the file to enable and disable certification. I assume this is left over from when i built the main_lctt_certif.c example. 

The only thing that seemed to work was to do a full flash erase and reprogram. This let certification to reset its state, and be disabled since I'm running periodical uplink example.

It seems counter-intuitive that this is necessary. Would it be possible to change this, so that the certification state is reset every time the device boots?

closed 2026-10-07 by lbm-team